earlephilhower/arduino-pico #2942: Feature Request: Added development board option for rp2xxx_stamp_carrier_xl_hw test board

mayjack0312, 2025-05-11:
Add [Test Board](https://github.com/solderparty/rp2xxx_stamp_carrier_xl_hw) content in the development board options

earlephilhower, 2025-05-12:
I took a quick look at that and it doesn't even have an RP2350 onboard.  It's just a breakout bare (well, no microprocessor) PCB for whatever other board you use.

You need select the appropriate base board since that would be where things like flash size and PSRAM would be defined.

closed 2025-05-12 by earlephilhower

mayjack0312, 2025-05-13:
> I took a quick look at that and it doesn't even have an RP2350 onboard. It's just a breakout bare (well, no microprocessor) PCB for whatever other board you use.
> 
> You need select the appropriate base board since that would be where things like flash size and PSRAM would be defined.

Thanks, it has been successfully run.