earlephilhower/arduino-pico #2958: Libraries not showing up in library manager

CorpusCallosum, 2025-05-15:
Hi, I followed the instruction to add the library json file in the Arduino setting:
https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json

But, when I search for Pico or Raspberry Pi or Philhower, the Raspberry Pi Pico library does NOT show up in the search results.

I only get these:

[image]
[image]

Any ideas?  Thanks.

earlephilhower, 2025-05-15:
This is a board package, not a library.  Assuming you followed the directions in https://github.com/earlephilhower/arduino-pico/blob/master/docs/install.rst#installing-via-arduino-boards-manager and put in the URL in the File->Preferences->Add'l Boards spot, then use `Tools->Board->Board Manager`, not `Sketch->Include->Manage Libraries`

closed 2025-05-15 by earlephilhower

CorpusCallosum, 2025-05-15:
Oh oops sorry about that! Thank you for the quick reply.