earlephilhower/arduino-pico #2960: Adafruit Metro RP2350 missing Tools > PSRAM selection  (file boards.txt missing menu.psram definitions)

PaulskPt, 2025-05-16:
I have reported this also in the Adafruit > # help-with-arduino channel on Discord.

Yesterday I received a new Adafruit Metro RP2350 that has 8Mb PSRAM (Adafruit product ID: 6267).

I tried to build and run the > Examples > rp2350 > PSRAMTest, 
however the Arduino IDE > Tools menu does not show the options to select "PSRAM Size ..." 
like it does when one choses the board: Adafruit Feather RP2350 HSTX.

So, the example sketch started reporting:
PSRAM Size: 0
followed by many lines like this one:
"Memory error @0x11000000 (0), was 0x00, should be 0xcb"

The file: 
C:\Users\<User>\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.5.3\boards.txt does not contain lines like:

```
adafruit_feather_rp2350_hstx.menu.psram.0mb=0MByte PSRAM
adafruit_feather_rp2350_hstx.menu.psram.0mb.build.psram_length=0x000000
adafruit_feather_rp2350_hstx.menu.psram.2mb=2MByte PSRAM
adafruit_feather_rp2350_hstx.menu.psram.2mb.build.psram_length=0x200000
adafruit_feather_rp2350_hstx.menu.psram.4mb=4MByte PSRAM
adafruit_feather_rp2350_hstx.menu.psram.4mb.build.psram_length=0x400000
adafruit_feather_rp2350_hstx.menu.psram.8mb=8MByte PSRAM
adafruit_feather_rp2350_hstx.menu.psram.8mb.build.psram_length=0x800000
```

In the Arduino IDE I have under > Preferences > Additional boards manager URLs: 
"https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json"

and in the BOARDS MANAGER:
Raspberry Pi Pico/RP2040/RP2350...  version 4.5.3 installed.

I tried to regenerate boards.txt using:
run 'python3 makeboards.py > ../boards.txt' to regenerate

however it stopped with reporting these errors:
Traceback (most recent call last):
  File "makeboards.py", line 485, in <module>
    pkgjson = json.load(open(os.path.abspath(os.path.dirname(__file__)) + '/../package/package_pico_index.template.json'))
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\<User>\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.5.3\\tools/../package/package_pico_index.template.json'

Platform on which I did this: desktop PC with MS Windows 11 Pro (version 24H2).

earlephilhower, 2025-05-16:
That board doesn't have any PSRAM onboard from the factory.  So, while the RP2350 chip does support it, unless you've manually soldered on a PSRAM SOIC then there really is no PSRAM...

PaulskPt, 2025-05-16:
Yes, that board is sold in two versions: without and with PSRAM. Mine  does have PSRAM. I bought it via Pimoroni Ltd. See [Metro RP2350 with PSRAM](https://www.adafruit.com/product/6267).
 Or [Pimoroni](https://shop.pimoroni.com/products/adafruit-metro-rp2350-with-psram?variant=55228529574267).

earlephilhower, 2025-05-16:
Ah, gotcha.  That model didn't pop up on Google for some reason.  In any case, they defined the CS pin in the header already so #2961 adds the IDE menus to let you select 0...8MB as the case may be.

PaulskPt, 2025-05-16:
You made me a happy guy! Have a nice weekend! Issue closed

PR #2961 (2025-05-16): Add PSRAM option for Adafruit Metro RP2350
Fixes #2960
closed 2025-05-16 by PaulskPt