earlephilhower/arduino-pico #2964: Feature request: Support BLE on RP2350 XRP

jhmaloney, 2025-05-21:
The new XRP robot controller board (https://www.sparkfun.com/sparkfun-experiential-robotics-platform-xrp-controller.html) is based on the RP2350 chip with the same CYW43 radio chip as the Raspberry Pi Pico2w. However, the pin connections to that chip are different on the XRP controller. They are defined in  variants/sparkfun_xrp_controller/init.cpp:

```
    static uint cyw43_pin_array[CYW43_PIN_INDEX_WL_COUNT] = {
        26, // CYW43_PIN_INDEX_WL_REG_ON
        29, // CYW43_PIN_INDEX_WL_DATA_OUT
        29, // CYW43_PIN_INDEX_WL_DATA_IN
        29, // CYW43_PIN_INDEX_WL_HOST_WAKE
        28, // CYW43_PIN_INDEX_WL_CLOCK
        27  // CYW43_PIN_INDEX_WL_CS
    };
```

The WiFi library is using those pin definitions but it appears that the Bluetooth stack is not. When I build and install MicroBlocks on a generic Pico2w board, I am able to use both WiFi and Bluetooth.

However, when I build and install the exact same code on the new XRP controller board, I wam able to use WiFi but not Bluetooth. I do not get any errors but neither BLE advertising nor BLE scanning works on the XRP controller.

I'm guessing that the CYW43 driver in the Bluetooth library is not using the pins defined in the XRP's cyw43_pin_array. I've been looking through the BTSTack source code but so far I haven't figured out where it is using those pins.

Hoping you can help. Thank you!

earlephilhower, 2025-05-21:
Also enable Tools->Debug Port->Serial and Tools->Debug Level->Bluetooth to get traces of HCI comms/status to look at...

earlephilhower, 2025-05-21:
Great, we'll see if we can find anything in the meantime.  And, first of all, can someone confirm that WiFi is working properly?  If that's not, then BT has no chance.  If it is working, then there is something subtle and weird...

jhmaloney, 2025-05-21:
I tried the KeyboardBT example and with both the XRP and a generic Pico2W I can see the board listed as a nearby keyboard device ("PicoW Password").

I can also confirm what WiFi works with the XRP board. (For anyone following this issue: you *must* set the board name to "sparkfun_xrp_controller" not "rpipico2w". Thanks to Dryw for setting me straight on that earlier.)

It looks like this is an issue in my own code, not a problem with the BTStack library. I'll investigate further when I have time; you don't need to spend any more time on this issue right now. I'll keep you posted...

earlephilhower, 2025-05-22:
If you find anything, drop a note here and we can reopen.  But I think we're all good...

jhmaloney, 2025-05-22:
I haven't had a chance to debug my own code, but I tried the BLE Peripheral example and that works as expected so I think we're good. Sorry for the false alarm -- and I really appreciate your responsiveness!

jhmaloney, 2025-05-24:
Hi, Earle. Just wanted to let you know that I found the bug in my code.

MicroBlocks resets all the GPIO pins INPUT mode on startup. That code was turning off the radio. Duh!

The MicroBlocks pin reset code skips the radio pins on the PicoW Pico2W boards, so BLE worked fine on those. However, the XRP controller uses different radio pins. Once I understood the problem, the fix was easy, of course.

Thanks again for being ready to help.