earlephilhower/arduino-pico #3025: Upload data to FS doesn't work on Pico2

KevinBDSL, 2025-07-06:
Hi

I've just started playing with the Pico2.  I'm having problems uploading data to the LittleFS data partition.  The New.UF2 file seems to get copied to the drive that appears when i trigger the command but the drive never closes and the pico2 doesn't reboot.  The whole thing just seems to hang up.

I'm using Arduino ide v1.8.19 as i don't get on with v2x

My sketch uploads correctly and will run but because the data files don't get uploadded the sketch does nothing )it looks for files to use).

I've tried my code and data on an original pico which works correctly.

It doesn't seem to matter what size files i put in the data folder or the size of FS i select (provided the files are smaller than this setting) it just does not work.

My last attempt i left the file explorer open for 30 minutes and nothing changed.

If i unplug my Pico2 from the usb and plug it in again my sketch runs but the FS is empty.

Any suggestions appreciated.

earlephilhower, 2025-07-06:
UF2s for the Pico and Pico 2 are not compatible.  The old Java plug in I did for 1.x doesn't know about the new format so it's not compatible. 

Only the [typescript version](https://github.com/earlephilhower/arduino-littlefs-upload) I wrote with with the Pico and Pico 2.  You need to use the 2.x IDE to do filesystem uploads on the Pico 2.

closed 2025-07-06 by earlephilhower

KevinBDSL, 2025-07-07:
OK
That worked.

Onwards and upwards.....

I think it would be useful to add this information to ReadTheDocs here 
https://arduino-pico.readthedocs.io/en/latest/fs.html#uploading-files-to-the-littlefs-file-system

and possibly here
https://arduino-pico.readthedocs.io/en/latest/rp2350.html

Kevin