earlephilhower/arduino-pico #3029: HTTP client drops data

KB1RD, 2025-07-07:
In reference to https://github.com/earlephilhower/BackgroundAudio/issues/61

I tried to get the background audio HTTP stream example to work, but its constantly stuttering without actually running behind. According to the error messages, there's large amounts of corruption in the received file.

I investigated by having the HTTP client print out the first three bytes of the file. I did not receive "ID3" as expected, so I investigated further. I modified the HTTP stream example as shown:

```cpp
/**
   StreamHTTPClient.ino

    Created on: 24.05.2015

*/

#include <Arduino.h>
#include <WiFi.h>
#include <HTTPClient.h>

#ifndef STASSID
#define STASSID "<redacted>"
#define STAPSK "<redacted>"
#endif

const char *ssid = STASSID;
const char *pass = STAPSK;

WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(11520000);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid, pass);
}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    Serial.print("[HTTPS] begin...\n");

    HTTPClient https;
    https.setInsecure();

    if (https.begin("https://path/to/my/mp3")) {

      Serial.print("[HTTPS] GET...\n");
      // start connection and send HTTP header
      int httpCode = https.GET();
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTPS] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK) {

          // get length of document (is -1 when Server sends no Content-Length header)
          int len = https.getSize();

          // create buffer for read
          static uint8_t buff[128] = { 0 };

          // read all data from server
          while (https.connected() && (len > 0 || len == -1)) {
            // get available data size
            size_t size = https.getStreamPtr()->available();

            if (size) {
              // read up to 128 byte
              int c = https.getStreamPtr()->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));

              // write it to Serial
              Serial.write(buff, c);

              if (len > 0) {
                len -= c;
              }
            }
            delay(1);
          }

          Serial.println();
          Serial.print("[HTTPS] connection closed or file end.\n");
        }
      } else {
        Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
      }

      https.end();
    } else {
      Serial.printf("Unable to connect\n");
    }
  }

  Serial.println("Wait 10s before the next round...");
  delay(10000);
}

```

And I read this onto my computer with:
```bash
(stty raw; cat > received.log) < /dev/ttyACM0
```

I trimmed the file in a hex editor so that it only displayed the MP3. This file *did* start with ID3 as expected, but there were massive differences. The audio stuttered on playback on my PC just as it did on the Pi.

It seems like the HTTP client is dropping characters every now and then. This likely has something to do with the larger flow rate since I'm able to maintain a websocket connection for long periods of time with no problem. It is worth noting: the server *will* attempt to send the whole file unless the Pi slows it down. Does the Pi not rate-limit the TCP connections, and instead discard the data?

earlephilhower, 2025-07-07:
Rate limiting is part of the TCP protocol.  There should be a limit to your server sending out packets without getting a TCP ACK for each one.  If it doesn't get an ACK or should resend the packet.  That's just the way TCP ensures a connection stream. 

What are you using to send the data to the Pico?  Are you using chunked HTTP?  A capture of the raw stream including headers would be useful.  The chunked handling was always a pain on the 8266 and maybe there's one more corner case in your config.

KB1RD, 2025-07-07:
I'll have to get back to you later but could I send you a link privately? Jellyfin doesn't seem to authenticate the user requesting a stream (other than to ensure the unique ID is stored on the server), and I don't want to publicly give out a URL that can be used to run transcoding on my server. Jellyfin has a demo server, but it does not used chunked encoding.

earlephilhower, 2025-07-10:
I was able to reproduce a failure using the [ChunkedClient.ino example](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/HTTPClient/examples/ChunkedClient/ChunkedClient.ino) here.  It doesn't always fail and when debugging is enabled it seems to fail less often.

Good output
````
<!DOCTYPE html>
<html lang=en>
<head>
<meta charset='utf-8'>
<title>Chunked transfer encoding test</title>
</head>
<body><h1>Chunked transfer encoding test</h1><h5>This is a chunked response after 100 ms.</h5><h5>This is a chunked response after 1 second. The server should not close the stream before all chunks are sent to a client.</h5></body></html>
Wait forever...
````

Same code, just reset the Pico
````
<!DOCTYPE ht<meta charset='utf-8'>
<title>Chunked transfer encoding test</title>
</head>
<body><h1>Chunked transfer encoding test</h1><h5>This is a chunked response after 100 ms.</h5><h5>This is a chunked response after 1 se
```

So there's some kind of race condition in the logic in `HTTPClient`'s chunked decoder.  It's grabbing the first few bytes of a chunk OK but sometimes actually discarding the rest of that chunk and jumping to the next one.  Very odd...

KB1RD, 2025-07-10:
The other problem is that when I'm playing audio, I get the wifi stream with `http.getStreamPtr()`, which goes around the chunked encoding -- Only the code in getString() passes through the decoding code. So, I need to write my own chunk decoder which proxies reads to the HTTP stream.

However, when I read out of the HTTP stream, it still starts with `f59`, which is nowhere in the stream, and is definitely not a chunk header. So, something is corrupting the underlying stream.

earlephilhower, 2025-07-10:
Yeah, it's an interesting design and not how I'd have done it, but I'm sure on the 8266 there were reasons for the choice.  Seems like a simple FSM wrapper like you did would be less code and simpler.  There were probably performance or memory constraints that aren't here now.

In any case, `getStreamPtr` should process the chunked encoding and there should not be that race condition.

FWIW the real mp3 web streaming apps don't use chunked, they just use a normal unterminated HTTP connection often with ICY blocks intersped every few KB.  So I never ran into that in BackgroundAudio.  That webserver is odd.  Apache doesn't do chunked encoding even when serving large MP3s in my limited testing...

PR #3034 (2025-07-11): Rewrite HTTPClient chunked transfer-encoding
Fixes #3029

The chunked decoder seems to have had some race conditions resulting in lost data.

The HTTPClient::getStreamPtr returned the raw WiFiClient which included all of the chunked block size markers, requiring the user to manually parse the chunked encoding.

Remove the existing chunked handling, add a HTTPStream as a WiFiClient subclass.  The HTTPClient will return this HTTPStream which will always properly cut out chunk markers when present (and pass things raw for non-encoded streams).  Use this new HTTPStream class to replace the existin [... cut]
commit f4de3cbd (2025-07-11): Rewrite HTTPClient chunked transfer-encoding (#3034)

* Rewrite HTTPClient chunked transfer-encoding

Fixes #3029

The chunked decoder seems to have had some race conditions resulting in
lost data.

The HTTPClient::getStreamPtr returned the raw WiFiClient which included
all of the chunked block size markers, requiring the user to manually
parse the chunked encoding.

Remove the existing chunked handling, add a HTTPStream as a WiFiClient
subclass.  The HTTPClient will return this HTTPStream which will always
properly cut out chunk markers when present (and pass things raw for
non-encoded stream [... cut]
closed 2025-07-11 by KB1RD