earlephilhower/arduino-pico #3058: Encancement: Support Custom CMake Flags and Selecting Targets in Build Script

schkovich, 2025-08-05:
Right now, the `make-libpico.sh` script does not let users pass custom CMake flags or specify build targets. This limitation makes it harder to adapt to local development and more complex continuous integration (CI) workflows.
## Motivation
Sometimes, developers only need to build specific targets. For instance, I have just the RP2040 Connect board, so it would be really helpful to compile only what I need to speed up my development process.

In addition, we often require custom CMake flags for different setups. For example, I use the Picoprobe and need a multicore build, which calls for some extra flags.
## Proposal
- Add support for positional arguments to specify build targets.
- Include a `--flags` option to allow for custom CMake flags to be forwarded to CMake.
## Expected Outcome
- Backwards compatibility will be maintained.
- Users will experience improved flexibility for both local development and CI builds.

earlephilhower, 2025-08-05:
I'm sorry, but this doesn't seem to have any use.  Users **never** need to build the library, it's a one-time thing when a developer updates the SDK.  In the worse case devs can push a PR to their private repo and run a GH action to rebuild the libraries in the cloud...it takes about 3 minutes to get the artifacts.

The example flags in the PR wouldn't actually do anything or be incompatible with this core.

We don't use `pico_stdio`, for example, so setting those `pico_stdio_xxx` would be no-ops.  We're already multicore and do not support the pico_multicore way of flash_safe_execute.  (FWIW, I implemented the flash lockout on my own here before the SDK had a way to do it and have stuck with this local implementation since it works with FreeRTOS and plain core without needing a SDK rebuild.)

Also, using picoprobe has no impact one way or the other on the SDK used.  It's just a debugger and can work with anything since it's running on a separate system + GDB.

Finally, the idea of libpico is that there's one version everyone using this core sees and it works (normally 😆) with the logical structure of this core.  This PR/enhancement seems to break that.

schkovich, 2025-08-06:
Yes, that is exactly what I am doing. I have an `async_context` instance on each core. The async context instance on core 0 handles interaction with LwIP, while the other one handles application logic. 

Check `async_context/include/pico/async_context_threadsafe_background.c`. For example, my favourite function. ;)

```
uint32_t async_context_threadsafe_background_execute_sync(async_context_t *self_base, uint32_t (*func)(void *param), void *param) {
    async_context_threadsafe_background_t *self = (async_context_threadsafe_background_t*)self_base;
#if ASYNC_CONTEXT_THREADSAFE_BACKGROUND_MULTI_CORE
    if (self_base->core_num != get_core_num()) {
        hard_assert(!recursive_mutex_enter_count(&self->lock_mutex));
        sync_func_call_t call = {0};
        call.worker.do_work = handle_sync_func_call;
        call.func = func;
        call.param = param;
        sem_init(&call.sem, 0, 1);
        async_context_add_when_pending_worker(self_base, &call.worker);
        async_context_set_work_pending(self_base, &call.worker);
        sem_acquire_blocking(&call.sem);
        return call.rc;
    }
#endif
    // short-circuit if we are on the right core
    lock_acquire(self);
    uint32_t rc = func(param);
    lock_release(self);
    return rc;
}
```

You will find [here](https://github.com/schkovich/pico-sdk-tests/tree/async-tcp) a test script designed to stress-test the async-tcp library's LwIP integration. It's a work in progress. The script is using QOTD and ECHO servers to simulate the real project requirements. The class `SerialPrinter` demonstrate how to use `async_context` to service libraries/functions that are not re-entrant or thread safe. 

It's running smoothly.

earlephilhower, 2025-08-06:
https://github.com/earlephilhower/arduino-pico/pull/3049/commits/26ca511900ed218faa4ce9262addef01a6576840 adds that define.

earlephilhower, 2025-08-07:
@schkovich if you could pull the latest commit to #3048 on your HW and test and report back I'd much appreciate it!

I've refactored things a lot and tried to unify the p.io build process with the main Arduino one (the defines never did match because they needed to manually be sync'd).  Make-LibPICO now also doesn't build and store 6 (BIG) copies of each LWIP option and instead just compiles on-the-fly in the IDE or P.IO.  Compiles in <1/2 the time and uses probably <1/2 the space of the prior releases.

I've been testing the Arduino side, but I don't(can't) run P.IO so I've just been checking CI doesn't get busted and things "look right."  A real test on 3rd party HW would be really nice to have.

schkovich, 2025-08-07:
[my branch](https://github.com/schkovich/pico-sdk-tests/tree/pico-sdk-2.2.0-update)

My local copy is on yetanothersdkupdate. 
```
git status
On branch yetanothersdkupdate
```
I did not build Pico SDK locally.

All environments compile and run without errors.

However, there is a significant shift in the test script's dynamics. I have to investigate. It might be trivial since everything is running locally on my desktop. It might be due to the removal of custom build flags. It might be related to "unify the p.io build process with the main Arduino one" refactoring related to LwIP. It might be something else. 

I want to add basic interactions with the board. For example, toggle the built-in LED on and off, and read data from the board's temperature sensor. If you have an idea how to increase the complexity further, let me know.

It might be a good opportunity to run the profiler. 

I will return with findings.

schkovich, 2025-08-07:
That was it. My stats are back to normal. 
```markdown
| Metric                     | Value        |
|----------------------------|--------------|
| QOTD connections           | 505          |
| Printed quotes             | 665          |
| Quote lines printed        | 1,995        |
| Total print ops            | 3,167        |
| Average print ops/sec      | 5.28/sec     |
| Quotes/connection ratio    | 1.32×        |
| Repeated quote prints      | 160          |
```
Before the latest update, I noticed a significant drop in the number of connections to the QOTD server. The number of connections dropped to around 120 per 10-minute run. 

I am going to add a blinking LED and temperature readings.

commit 2de633b5 (2025-08-07): Update to Pico-SDK 2.2.0 release, remove multi-lib LWIP, clean up CMake, clean up P.IO build hardcoded defines (#3049)

* Update to Pico-SDK 2.2.0 release

Doesn't seem to have many additions since the develop branch we were using,
but this one is nice and official and tagged.

* Add LIB_PICO_MULTICORE to libpico build
Fixes #3058

* Clean up some unused param warnings

* Fix olimex XL/XXL RP2350A redefinition

* Clean up SDK defines in IDE builds

* Build LWIP at app compile time, no need for multi

We needed about 6 copies of the LWIP/BT library for each CPU when the
LWIP library was built a [... cut]