earlephilhower/arduino-pico #3062: Support Pimoroni Pico Explorer

libewa, 2025-08-08:
The Pimoroni Pico Explorer is a development board featuring with the RP2350 controller, 6 on-board buttons, 4 servo outputs, a TFT screen and a small breadboard. It would be great to control these from the Arduino environment.

libewa, 2025-08-08:
Dear god. Documentation on this board SUCKS. The micropython repository is simply a copy of PGA2350's, there's no pinout sheet...

There's supposedly 5 ADCs on this. Even the Pico Plus 2 only defines 2.

I'll check explorer.py... once my 3283 JPEGs are copied over 11500 UART.

earlephilhower, 2025-08-08:
FWIW, the RP2040/RP2350A chip itself has 4 ADC channels + 1 temperature, so 5 ADCs in total if you want to look at it that way.  The RP2350B actually has 8 + 1 channels.

libewa, 2025-08-09:
It's an RP2350B.
I actually found a schematic: <https://cdn.shopify.com/s/files/1/0174/1800/files/Pimoroni_Explorer_Schematic.pdf?v=1729171995>, but am unsure how to represent all of this IO.

Should `PINS_COUNT` be the pins on the board or chip? How do I use I2C 2? Servos? The amplifier? Is the display in scope for the core?

libewa, 2025-08-09:
[image]

IT'S ALIVE!
And my filesystem survived.

libewa, 2025-08-11:
One more thing: Is it possible to add additional documentation, like how to use the screen with Adafruit GFX?

PR #3067 (2025-08-11): Support the Pimoroni Explorer board.
Fixes #3062
closed 2025-08-11 by earlephilhower