earlephilhower/arduino-pico #3068: SPISlave seems to crash the controler

siteswapjuggler, 2025-08-09:
Hello, 

I'm really sorry to bother you but I'm running out of ideas. I'm new to Pi Pico but quiet experience with ESP32 and AVR boards. I'm trying to get an ESP to send data via I2C and/or SPI  a Pi Pico.

I2C works well no problem, but with SPI i'm totally blocked. The Pico seems to start but crash as soon as the SPISlave.begin is called then not responding anymore, no more Serial, the loop doesn't run... just stuck.

I read the other issues, so tried to change SPI_MODE, also tried to add external pull up to test for electronical causes wihout any success. Also tried on another Pico and a Pico 2 and same result with this code the controller get stucked at SPISlave.begin. 

`
#include <SPI.h>
#include <SPISlave.h>

const int LED_PIN = LED_BUILTIN;

const int PIN_CS = 17;
const int PIN_SCK = 18;
const int PIN_RX = 16;
const int PIN_TX = 19;

SPISettings spisettings(1000000, MSBFIRST, SPI_MODE0);

void onRecv(uint8_t *data, size_t len) {
  if (len == 1) {
    analogWrite(LED_PIN, data[0]);
  }
}

void onSent() {
}

void blink(int n, int d = 150) {
  while (n--) {
    digitalWrite(LED_PIN, 1);
    delay(d);
    digitalWrite(LED_PIN, 0);
    delay(d);
  }
}

void setup() {
  Serial.begin(115200);

  pinMode(LED_PIN, OUTPUT);
  analogWriteFreq(1000);
  analogWriteRange(255);

  SPISlave.setRX(PIN_RX);
  SPISlave.setCS(PIN_CS);
  SPISlave.setSCK(PIN_SCK);
  SPISlave.setTX(PIN_TX);

  SPISlave.onDataRecv(onRecv);
  SPISlave.onDataSent(onSent);

  blink(1, 120);  // marker #1: before slave begin
  // the program stop here.................
  SPISlave.begin(spisettings);
  blink(2, 120);  // marker #2: after slave begin

  Serial.println("SPI slave ready");
}

void loop() {
  Serial.println("Hello");
  analogWrite(LED_PIN, 0);
  delay(1000);
  Serial.println("World");
  analogWrite(LED_PIN, 64);
  delay(1000);
}
`

earlephilhower, 2025-08-09:
@siteswapjuggler thanks for the MCVE.  I verified the existing SPIToMyself.ino sketch still worked than then dived into your code.  The difference is you never send data back, so the TXFIFOEMPTY IRQ gets asserted but never cleared.

Please try #3069 and report back.  Locally I can see it seems to work (but I don't have anything speaking SPI to the board to test recv. functionality).

siteswapjuggler, 2025-08-10:
Hi thanks a lot for your answer I'll try that this afternoon and let you know the result :)

siteswapjuggler, 2025-08-10:
Hi @earlephilhower, just tested it right now and it work like a charm :) thank a lot for your help.

commit 2f983dee (2025-08-10): Handke SPISlave with no output data ever written (#3069)

Fixes #3068.  The SPISlave TX-FIFO IRQ would infinitely try to fill the FIFO
as soon as IRQs were enabled, but in the case when the app never sent any
data this would cause an infinite loop in the IRQ handler.

Now abort if we don't have any data, even if there is space in the FIFO.
closed 2025-08-10 by siteswapjuggler