earlephilhower/arduino-pico #3072: Add support for (Chinese?) RP2040 Pico Uno boards

waltje, 2025-08-12:
These are now floating about, in two basic variants: one uses the same Wifi module as the PicoW, so should be easy as it apparently is pretty much the same schematic but on a larger board footprint, and the other one seems to have an ESP-12 chip mounted (ESP8266) to implement WiFi.

I have one of the former, and one of the latter is on its way for testing with.

[image]

earlephilhower, 2025-08-12:
Is that really a full size USB-B connector in 2025?  Ouch!

Anyway, really old printer cables aside, it looks like it's got some extra peripherals (ESP32 WIFI?) so you might be the best person to add it.  See the docs [here](https://github.com/earlephilhower/arduino-pico/blob/master/docs/contrib.rst#adding-a-new-board) which describe the process.  Assuming you're using an ESP32 that's supported, and can flash that WiFi chip, then you should even be able to get [ESPhost](https://github.com/earlephilhower/arduino-pico/tree/master/libraries/lwIP_ESPHost) running to give full internet connectivity.

waltje, 2025-08-12:
Yes, it is :)  I actually still like those.. not as frail as the mini/micro/c ones.

The board uses an ESP-12 (aka 8266EX) for Wifi.  I do not yet have the schematics (it is based
on a German design for a board with the same name (the Eins); this is a Chinese clone) but I do
not see a Flash chip near the ESP, so I am somewhat surprised.  I dont think one can blast firmware
into its RAM on the fly, so how... ?

Abyway, first step would be to get the basics up and running.

earlephilhower, 2025-08-12:
You're unfortunately out of luck with Wifi, then.  I bet they saved $0.05 in going with an 8266 over a ESP32-S2 with 2x the CPU power, 5x the RAM, and probably 10x the network performance. 😭 

ESPHost only runs on ESP32-class chips (not all, but many).  The 8266 is probably stuck running a *really* bad `AT`-interface firmware.   You'd communicate over one of the UARTs on the Pico.  There are Arduino libraries to do simple things, like send a request to a plain HTTP server (no HTTPS) to post a value and stuff, if you look for them.

Some ESP8266 devices had flash in the package, maybe that's what they're using.

In any case, the Pico side is going to be very simple and you'll probably just need to select the right UART pins and LED connection to get it running.

dquadros, 2025-08-13:
Just bought one out of curiosity, should be here in 2 or 3 weeks. The ad says it is an ESP8285, that is an ESP8286 with integrated Flash (1 ou 2M). You probably can load a custom software in it for better performance (the UART can run up to 4Mbps), but does not look worth the effort. The should have gone with an ESP32-Cx for a better cost/performance (a C2 or C3 would add BLE, a C6 would add BLE and Zigbee/Thread).

waltje, 2025-08-14:
Good luck.  I have tried everything to get a schematic for it - none of the sellers can give them.  One actually said "We cannot do that", so they may or may not have them.  I *think* I figured out the actual manufacturer, so it may still happen.

As a basic board, this should be workable, by starting with the Eins board definition, and probe all pins to make sure they are the same, then checking out the LED pin.  A blinky for the Eins does NOT blink on this one, so, looks like different pin def.

The WiFi part... I don't think I am going to even try.

waltje, 2025-08-14:
> See the docs [here](https://github.com/earlephilhower/arduino-pico/blob/master/docs/contrib.rst#adding-a-new-board) which describe the process.

```
Traceback (most recent call last):
  File "/usr/local/arduino/portable/packages/rp2040/hardware/rp2040/4.6.0/tools/makeboards.py", line 489, in <module>
    pkgjson = json.load(open(os.path.abspath(os.path.dirname(__file__)) + '/../package/package_pico_index.template.json'))
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/arduino/portable/packages/rp2040/hardware/rp2040/4.6.0/tools/../package/package_pico_index.template.json'
```

Not good.  This is on Linux, but that should matter.

earlephilhower, 2025-08-14:
Make sure you're running from a full `git clone` of the repo, not the installed version from the IDE.

waltje, 2025-08-14:
> Make sure you're running from a full `git clone` of the repo, not the installed version from the IDE.
Hmm, yeah, ok, that makes sense :P

waltje, 2025-08-14:
Well, this is not pretty.

I manually "pinged through" all board pins.  The board actually has screwed-up pin connections:

0..3 are OK, connected to D0..3
4,5 are not connected
6..11 are OK, but connected to D4..9
12,13 not connected
14,15 OK, connected to A4,5 (I2C)
16 = D12
17 = D10
18 = D13
19 = D11
20..24 not connected
25 = LED
26..29 connected to A0..3

So, this is messy.  I was able to re-code my test sketch with "updated" pin numbers, and, yes, that
does work.

What does NOT work, is SPI.  Obviously, we have a problem here, because the SPI pin numbers are
totally messed up.

Is there a way an internal "translation table" can be used to fix this up inside the core, so that the
"pin numbers" (0..25) match the board pins again, AND for SPI's internal pins?

waltje, 2025-08-14:
> Just bought one out of curiosity, should be here in 2 or 3 weeks.
See above. You might want to cancel that ;-)

> The ad says it is an ESP8285, that is an ESP8286 with integrated Flash (1 ou 2M).
Yes, it is the 1MB variant of it.

I guess one could write some simple-ish firmware for it to handle IPv4, possibly based
on one that we already have, but hrrm.

I did add the board to the core, in a non-Wifi and a Wifi variant, assuming the latter one
may never come alive.