earlephilhower/arduino-pico #3074: BluetoothHIDMaster KeyboardPiano.ino won't connect to Bluetooth Classic keyboards

jedgarpark, 2025-08-14:
I'm attempting to connect various Bluetooth Classic keyboards with BluetoothHIDMaster > KeyboardPiano.ino and having no luck. I put them in pairing mode but no connection is made. Arduino tools settings are IPv4 + Bluetooth and USB stack is PicoSKD

They do show up in the scanner example (BluetoothHCI > BluetoothScanner):

```
BEGIN SCAN @21402 ...END SCAN @25253
Class    | Address           | RSSI | Name
-------- | ----------------- | ---- | ----------------
00002540 | dc:2c:26:f6:b0:98 |  -58 | Keychron K8
00002540 | 70:f0:87:27:2b:e7 |  -40 | Magic Keyboard
00002540 | e4:17:d8:f0:3a:aa |  -32 | 8BitDo Micro gamepad
```
So they're being seen, but just don't want to connect.

Using the BluetoothHIDMaster > KeyboardPianoBLE.ino example and a different keyboard that is BLE works great.

Have tried on PicoW and Pico 2W and Pimoroni Pico Plus 2W. 

Pico board package is 4.7.1, Arduino IDE is 2.3.6.

earlephilhower, 2025-08-14:
This is actually by design.  I probably should have named it "BLEHIDMaster" or something but the current setup is a BLE HOG (HID over GATT) master and not Classic BT.

I have maybe a half dozen HID keyboards and mice in the house, and all are BLE so I didn't imagine there were classic BT-based ones (BLE is way better for battery powered, low bandwidth devices like mice and keyboards).  BT and BLE are both "Bluetooth" but completely different WRT the underlying interface and APIs.  It's absolutely possible to write something to be a classic BT HID master, but it would not share much with the existing BLE implementation.  (The HID processing would be identical, of course, since everyone is speaking USB HID 😆 )

I'd be happy to take a PR that implements classic BT HID master, but since I don't have anything to test it's not something I have in the cards, sorry!

closed 2025-08-14 by earlephilhower

jedgarpark, 2025-08-14:
@earlephilhower right on, no worries. I think BT Classic shows up more on devices that are concerned with latency such as gamer keyboards and game controllers? I was able to get this working using BTStack, here's example code in case it's useful to anyone else:
`// Bluetooth Classic HID Connection for Pico W
// Step 1: Add HID connection framework to working discovery

extern "C" {
#include "btstack.h"
}
#include <BluetoothHCI.h>

BluetoothHCI hci;

// LED status
unsigned long ledTimer = 0;
bool ledState = false;

// HID Connection state
typedef enum {
    HID_DISCONNECTED,
    HID_CONNECTING,
    HID_CONNECTED
} hid_connection_state_t;

hid_connection_state_t hid_state = HID_DISCONNECTED;
bd_addr_t target_keyboard_addr;
bool target_keyboard_found = false;

// L2CAP channel IDs for HID
uint16_t hid_control_cid = 0;
uint16_t hid_interrupt_cid = 0;

// BTStack packet handler
static btstack_packet_callback_registration_t hci_event_callback_registration;

void setup() {
  Serial.begin(115200);
  delay(5000);
  
  Serial.println("=== BLUETOOTH HID CONNECTION DEMO ===");
  Serial.println("Step 1: Discovery + Connection Framework");
  Serial.println("Starting Bluetooth initialization...");
  
  // Initialize LED
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  
  // Initialize HCI
  l2cap_init();
  sm_init();
  gap_set_default_link_policy_settings(LM_LINK_POLICY_ENABLE_SNIFF_MODE | LM_LINK_POLICY_ENABLE_ROLE_SWITCH);
  hci_set_master_slave_policy(HCI_ROLE_MASTER);
  hci_set_inquiry_mode(INQUIRY_MODE_RSSI_AND_EIR);

  hci.install();
  hci.begin();
  
  // Register BTStack event handler for HID connection
  hci_event_callback_registration.callback = &packet_handler;
  hci_add_event_handler(&hci_event_callback_registration);
  
  // Turn on Bluetooth
  hci_power_control(HCI_POWER_ON);
  
  Serial.println("HCI initialized. Starting first scan...");
  scanForDevices();
}

void scanForDevices() {
  if (hid_state == HID_CONNECTED) {
    Serial.println("Already connected to HID device. Skipping scan.");
    return;
[... 238 lines cut ...]
  // Auto-scan every 20 seconds if not connected
  if (hid_state == HID_DISCONNECTED && (millis() - lastScan >= 20000)) {
    scanForDevices();
    lastScan = millis();
  }
  
  // Manual scan on BOOTSEL press
  if (BOOTSEL) {
    while (BOOTSEL) {
      delay(1);
    }
    Serial.println("BOOTSEL pressed");
    if (hid_state == HID_CONNECTED) {
      Serial.println("Disconnecting from HID device...");
      l2cap_disconnect(hid_control_cid);
      l2cap_disconnect(hid_interrupt_cid);
    } else {
      Serial.println("Manual scan triggered");
      scanForDevices();
    }
    lastScan = millis();
  }
  
  delay(100);
}`

earlephilhower, 2025-08-14:
Nice work!  Looks like you're pretty close to having something general purpose.  Maybe you'd like to see about working that into the BluetoothHIDMaster as a partner class or something?  Things like the HIDKeyStream can work with the reports you're getting through BT classic HID and make it look like any other `Stream` for apps.

jedgarpark, 2025-08-15:
@earlephilhower sounds great! I'm going out of town for a few days, I'll look at this next week.

earlephilhower, 2025-08-16:
So actually as I was going through some of the network code while refactoring for FreeRTOS I realized that my original message was bogus.  Been a while since I looked at it but obviously I only say the GATT bit and not the BT classic later in the code.  🤯 

AFAIK BluetoothHIDMaster supports BLE and classic BT.  Calling `begin(false` parameter should use the classic path.  If not, we have a big in existing code.

I don't have any BT classic keyboards, but I do think I have a 8bitdo pad which IIRC is BT classic and can always fire up a PicoW to pretend to be a BE classic keyboard.  If I get some time I'll fire them up and see what's happening in the code here.  Might be something as simple as a class ID or something mismatch...

earlephilhower, 2025-08-20:
I just ran the PicoW keyboard demo and was able to repro the issue.  It's not a problem in the library, but in the demo.  If the initial scan doesn't succeed (which can happen if BTStack doesn't see the keyboard during the scan window) we should be attempting to re-scan.  The demos were just updated to do so, and now I can connect to the BT HID Master example using the PicoW BTKeyboard demo.

PR #3089 (2025-08-21): Rescan if first one fails in HIDMaster examples
Sometimes the BT or BLE announcement doesn't show up during an initial scan window.

Loop in the KeyboardPiano examples so that scans will be repeated until something actually links up.

Fixes #3074