earlephilhower/arduino-pico #3131: __wrap___stack_chk_fail@0x10017090 main.cpp243:0

drmpf, 2025-09-13:
Using Arduino IDE V2.3.6
with arduino-pico board support V5.1.0

Programming Pi Pico W via Pi Pico 2020 
Compile and debugging example WiFiServer.ino 
// Placed in the public domain by Earle F. Philhower, III, 2022

With settings

[image]

gives compile output
```
Sketch uses 429552 bytes (41%) of program storage space. Maximum is 1044480 bytes.
Global variables use 76752 bytes (29%) of dynamic memory, leaving 185392 bytes for local variables. Maximum is 262144 bytes.
```

Debugging with break point at line 19 WiFi.mode(WIFI_STA);
halts with call stack
```
_exit@0x1002dfb6
newlib_interface.c45:0
panic@0x1001aaaa
__wrap___stack_chk_fail@0x10017090
main.cpp243:0
runtime_init@0x1002e05a
newlib_interface.c200:0
platform_entry@0x1000310e
```

and debug output
```
Reading symbols from C:/Users\matthew\AppData\Local\arduino\sketches\1771BB55525B5A5F0E2BE8F5325A47D2\WiFiServer.ino.elf...
Finished reading symbols from objdump: Time: 255 ms
Finished reading symbols from nm: Time: 268 ms
warning: could not convert '$ITM_BASE' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
0x1002ef82 in ?? ()
Program stopped, probably due to a reset and/or halt issued by debugger
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
Note: automatically using hardware breakpoints for read-only addresses.
Thread 
1 "rp2040.core0" received signal SIGTRAP, Trace/breakpoint trap.
_exit (status=status@entry=1) at C:\Users\matthew\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\5.1.0\cores\rp2040\sdkoverride\newlib_interface.c:45
45	        __breakpoint();
```

Trying to step over or continue results in the same stack trace. 
The Program will not run.

Removing the breakpoint gives same result, the program will not run.

maxgerhardt, 2025-09-13:
Does this occur when Tools -> Optimize is set to Os, O2, O3 as well? Or only on Og or O0?

earlephilhower, 2025-09-13:
These traces show something in the SDK runtime startup is, or at least looks to stack smashing checks, blowing the stack. It's before our code here even runs (before main() is called).

maxgerhardt, 2025-09-13:
When you upload the sketch normally, does it run? `WiFiServer.ino` not running anymore on a Pico W would be quite severe.

drmpf, 2025-09-14:
Thanks.  
I off both Stack Protections and C++ exceptions, since I don't throw anything myself.  
Examples debug fine now.

earlephilhower, 2025-09-25:
This was actually an interesting issue that was simple once I let it sit a bit.

The problem is that `runtime_init` is called on machine startup and it iterates thru a list of pre-C-library-runtime initializers (for things like auto_init_mutexes, etc.).  One of those pre-C-lib init routines is the function which actually enables stack smash protection.

https://github.com/earlephilhower/arduino-pico/blob/f62e680980abf9a1c645266481ab38832b16930a/cores/rp2040/sdkoverride/newlib_interface.c#L194-L199

So runtime_init starts with stack protection off, sets it on in the middle of the iterator, and then returns...but GCC inserts a "check stack wasn't busted" call on that return.  And since the function started w/o setting up the canary, when it's checked for, it's not there and `panic` happens.

It's a trivial fix, just turning off stack protection for the runtime_init function with an `attribute`