earlephilhower/arduino-pico #3152: RP2040 Crash with Serial Disconnect in 5.2.0

OGMAvision, 2025-09-25:
My code was working using core 5.1.0.

I have just upgraded to Core version 5.2.0, and my code is no longer working. This is the only change I have made.

In the IDE, I have selected the new upgraded core. 

My board is a custom board based on Raspberry Pi Pico (Earle Philhower core).

My IDE is VSCode with Arduino Maker Workshop extension.

**Symptoms**

The device crashes as soon as it starts working, and the USB serial (COM12) disconnects. After it disconnects, it becomes impossible to reconnect until I reset the board.

The crash happens very early, and I haven't been able to produce any meaningful log.

**Code**

At the top of my dot INO sketch, I have:

```
#include "config.h"
#include "debug.h"
#include "debug_ram.h"
#include "motor_accelstepper.h"
#include "sensor.h"
#include "serial_cmd.h"
#include "controller.h"

void setup() {
  Serial.begin(115200);
  delay(2000); // Give Serial connection time to establish
}
```

I saw the following in the Change Logs:

"NOTE: With this change, devices don't register on the USB port until you ::begin them. Make sure you call Keyboard.begin(), etc. as appropriate in your setup if you're not doing so already."

I believe that having the line `Serial.begin(115200);` is sufficient to meet the above requirement. My code already had that line since the previous version of the core. Am I missing something?

SatreDaniel, 2025-09-25:
Raspberry Pi Pico W 
Programme de communication RS232, ModBus, petit automate et stockage sur SD Card
Idem, blocage total dès le démarrage, j'ai supprimeé la SD-Card le RS485 modifié le RS232 rein ni fait

earlephilhower, 2025-09-25:
I don't have your custom motor libs, so to compile I removed them and ran
````
void setup() {
  Serial.begin(115200);
  delay(2000); // Give Serial connection time to establish
}

void loop() {
  // put your main code here, to run repeatedly:
Serial.println(42);
}
````
on a Pico 2 just now under Linux and Win 11.  No issue, it prints a stream of "42"s.

> I believe that having the line Serial.begin(115200); is sufficient to meet the above requirement. My code already had that line since the previous version of the core. Am I missing something?

No, that's not correct.  It's only if you are changing the USB descriptor and then only if you're using the low-level USB.xxx calls, not one of the packaged libs.  SerialUSB handles this already, and in fact is called before `setup` to enable the reset mechanism.  The call in `setup` is a no-op because `SerialUSB` has already started
https://github.com/earlephilhower/arduino-pico/blob/512b2918f82466d8b6542936414d591f6359df37/cores/rp2040/SerialUSB.cpp#L45-L62

However, running on a PicoW I did see your problem and GDB showed
````
#2  0x10004564 in noInterrupts () at /home/earle/Arduino/hardware/pico/rp2040/cores/rp2040/wiring_private.cpp:83
83	        panic("IRQ stack overflow");
(gdb) list
````

It's related to the IRQ stack optimization I threw in at the last minute to save 128 bytes of heap.  I forgot to decrement the IRQ stack count and the CYW43 has lots of interrupt disable/enables, causing it to think it overflowed.

#3153 re-adds the missing decrement and should fix your problem.

PR #3153 (2025-09-25): Restore missing irqStackTop decrement
Fixes #3152