earlephilhower/arduino-pico #3156: error when defining SPI for the W5500.

JacTech, 2025-09-27:
I revisited a project from a few months ago, sadly the code does no longer compile. 
The error seems to be with the definition of SPI1 for the W5500 chip
```c++
Wiznet5500lwIP eth(13, SPI1);
```
here is the terminal output
```
src\main.cpp:24:28: error: no matching function for call to 'LwipIntfDev<Wiznet5500>::LwipIntfDev(int, SPI0_Type*)'
   24 | Wiznet5500lwIP eth(13, SPI1);
      |                            ^
In file included from C:\Users\Jac\.platformio\packages\framework-arduinopico\libraries\lwIP_w5500\src/W5500lwIP.h:3,
                 from src\main.cpp:4:
C:\Users\Jac\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet\src/LwipIntfDev.h:75:5: note: candidate: 'LwipIntfDev<RawDev>::LwipIntfDev(int8_t, arduino::SPIClass&, int8_t) [with RawDev = Wiznet5500; int8_t = signed char; arduino::SPIClass = arduino::HardwareSPI]'
   75 |     LwipIntfDev(int8_t cs = SS, SPIClass& spi = SPI, int8_t intr = -1) :
      |     ^~~~~~~~~~~
C:\Users\Jac\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet\src/LwipIntfDev.h:75:43: note:   no known conversion for argument 2 from 'SPI0_Type*' to 'arduino::SPIClass&' {aka 'arduino::HardwareSPI&'}
   75 |     LwipIntfDev(int8_t cs = SS, SPIClass& spi = SPI, int8_t intr = -1) :
      |                                 ~~~~~~~~~~^~~~~~~~~
C:\Users\Jac\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet\src/LwipIntfDev.h:73:7: note: candidate: 'constexpr LwipIntfDev<Wiznet5500>::LwipIntfDev(const LwipIntfDev<Wiznet5500>&)'
   73 | class LwipIntfDev: public LwipIntf, public RawDev {
      |       ^~~~~~~~~~~
C:\Users\Jac\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet\src/LwipIntfDev.h:73:7: note:   candidate expects 1 argument, 2 provided
C:\Users\Jac\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet\src/LwipIntfDev.h:73:7: note: candidate: 'constexpr LwipIntfDev<Wiznet5500>::LwipIntfDev(LwipIntfDev<Wiznet5500>&&)'
C:\Users\Jac\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet\src/LwipIntfDev.h:73:7: note:   candidate expects 1 argument, 2 provided
```

This also happens when defining SPI0, the only way i could "resolve" it is by just defining the CS pin and no SPI.

earlephilhower, 2025-09-27:
I can't reproduce that here.  All constructor variants build without error with the following sketch on the latest core:

````
#include <W5500lwIP.h>
Wiznet5500lwIP eth(17, SPI1);
Wiznet5500lwIP eth2(17, SPI, 21);
Wiznet5500lwIP eth3(17);

void setup() {
}

void loop() {
}
````

Please provide a full MCVE that fails so we can look into it.

earlephilhower, 2025-09-28:
Also, there is no `SPI0_Type` in this core (what GCC's saying you've passed in to the 2nd param of the constructor.) 

````
src\main.cpp:24:28: error: no matching function for call to 'LwipIntfDev<Wiznet5500>::LwipIntfDev(int, SPI0_Type*)'
````

 You might be using some third party library implementing a weird SPI object that's not compatible with the normal Arduino SPI classes.

maxgerhardt, 2025-09-28:
Since you're using PlatformIO: I have just *personally* verified that the above code statement compiles normally.

[image]

That is, taking https://github.com/earlephilhower/arduino-pico/blob/ded8a8a8d9a83f1755e47bbed05c619cdb8ad14c/libraries/lwIP_w5500/examples/WiFiClient-W5500/WiFiClient-W5500.ino and [this block](https://github.com/earlephilhower/arduino-pico/blob/ded8a8a8d9a83f1755e47bbed05c619cdb8ad14c/libraries/lwIP_w5500/examples/WiFiClient-W5500/WiFiClient-W5500.ino#L11-L16) with 

```cpp
Wiznet5500lwIP eth(13, SPI1);
```
results in a perfect compilation with 0 warnings and 0 errors. 

You can make sure platform-raspberrypi is up-to-date by deleting the `C:\Users\<user>\.platformio\platforms\raspberrypi*` folder(s) and building the project anew, just making sure the `platform` points to the URL then
```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
```

JacTech, 2025-09-28:
Tanks for the quick responses, the error seems to have come from including ```<rp2040.h>```
When i was trying to replicate the error in a MCVE i couldn't, however i noticed ```SPI1``` linked to a different definition, here it was
```.platformio > packages > framework-arduinopico > libraries > SPI > src > C SPI.h > [e] SPI1"```
while in the code with the error it was
```.platformio > packages > framework-arduinopico > pico-sdk > src > rp2_common > cmsis > stub > CMSIS > Device > RP2040 > Include> C RP2040.h> E SPI1"```
Removing the ```#include <rp2040.h>``` resolved the issue and the code works again. Don't ask me why it was included in the first place  ¯\_(ツ)_/¯
Anyways thanks for the responses and sorry for the time waste.

closed 2025-09-28 by earlephilhower